technoblogy/attiny10core #4: Non-functional on Arduino IDE 1.8.9

goobering, 2019-08-11:
I'd love to use this. Unfortunately it's not working with a fresh install of Arduino IDE 1.8.9.

Per the instructions, I've put the extracted zip file in a freshly created hardware folder in my Documents\Arduino folder. After that I can see (and have selected) the "ATTiny10/9/5/4" option in my Boards menu, and have selected "USBasp" as my programmer. When I try to build/verify the blink sketch provided [here](http://www.technoblogy.com/show?1YQY) the IDE errors out with `Fatal error: unknown MCU: avrtiny`. I *think* the problem might be related to the AVR-GCC version included with the Arduino IDE, but I'm still trying to test that. Any suggestions very welcome.

technoblogy, 2019-08-12:
OK, I'll check it myself and let you know.

technoblogy, 2019-08-14:
Are you running the Arduino IDE on Linux?

goobering, 2019-08-14:
I'm afraid not - Windows 10 for me. I managed to upgrade the AVR-GCC install and have made a little progress, but I hit outdated firmware on my USBAsp programmer - going to try to get that updated using another Arduino later tonight and I'll see where that takes me.

technoblogy, 2019-08-15:
That could be the problem - you need the latest firmware. Let me know how you get on.

technoblogy, 2021-02-28:
These issues should now be solved by the new Boards Manager installation. Thank you for your suggestions.

closed 2021-02-28 by technoblogy