technoblogy/attiny10core #5: incompatable with 1.8.10

uggima, 2019-12-23:
firstly, thanks for the great work and not expecting you to fix this just thought i'd report it.

was attempting to use your core after a fresh install of windows and any compilation that didn't result in an error (including PUEB is undefined) returned:
`
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: cannot find crtattiny10.o: No such file or directory

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: skipping incompatible c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib\libm.a when searching for -lm

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: cannot find -lm

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: skipping incompatible c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0\libgcc.a when searching for -lgcc

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: cannot find -lgcc

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: skipping incompatible c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib\libm.a when searching for -lm

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: cannot find -lm

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: skipping incompatible c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib\libc.a when searching for -lc

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: cannot find -lc

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: cannot find -lattiny10

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ATtiny10/9/5/4.`

reverting to 1.8.8 solved the issue

technoblogy, 2019-12-24:
Thanks for reporting this - I'll see if I can figure out what's changed. David

technoblogy, 2019-12-30:
Bit more about this here:

https://www.avrfreaks.net/forum/error-cannot-find-crtattiny10o-no-such-file-or-directory

Reverting to Arduino IDE 1.8.8 is currently the best workaround.

settinger, 2020-09-19:
It appears to be an error with AVR-gcc. The compiler toolchain in 1.8.8 and earlier uses AVR-gcc version 5.4.0; 1.8.9 and later use AVR-gcc version 7.3.0. I solved the issue by further upgrading to AVR-gcc version 10.1.0 using the instructions provided here: https://blog.zakkemble.net/avr-gcc-builds/

SpenceKonde, 2020-12-17:
If this core can be installed via board manager, you can specify the version of avr-gcc to use. You say 5.4.0, and specify the tarballs arduino used for 1.8.8 and earlier? It'll go download those and use them to compile...

chovanj, 2021-02-14:
> It appears to be an error with AVR-gcc. The compiler toolchain in 1.8.8 and earlier uses AVR-gcc version 5.4.0; 1.8.9 and later use AVR-gcc version 7.3.0. I solved the issue by further upgrading to [...]

**I've just tested it and it works and attiny10 programming works perfectly again. The Arduino IDE also looks building fine my sketches.**

Why don't you upgrade The Arduino IDE also to 10.1.0 toolchain?

technoblogy, 2021-02-14:
> Why don't you upgrade The Arduino IDE also to 10.1.0 toolchain?

Sorry, I'm not quite sure how I could do this - any suggestions?

SpenceKonde, 2021-02-14:
I absolutely would not go upgrading it to 10, lol....Not without a huge amount of testing to convince myself that it was actually better. 

One doesn't need to upgrade the toolchain that the whole damned IDE uses. This could be fixed very easily if this core was converted to install via boards manager, because that manages tools_dependencies precisely for this reason. I'm not sure why he hasn't made a board manager package out of this!

Once it was made into board manager package, it could just declare that it required that working 5.4.0 release (wouldn't even need to say where to get it, just that it was packaged by arduino and the version string and it would then look at their json file, find the url to download the right version for your OS from, install it, and then install this core and you wouldn't risk breaking other things that depended on the 7.3 version, and nobody would have to have and older version of the IDE installed to use your core, (and the latest version for everything else - 1.8.13 has some user experience improvements around third party hardware packages (the menus) that are just amazing....

chovanj, 2021-02-14:
> > Why don't you upgrade The Arduino IDE also to 10.1.0 toolchain?
> 
> Sorry, I'm not quite sure how I could do this - any suggestions?

I mean how this guy does that https://blog.zakkemble.net/avr-gcc-builds/
If anyone uses AVR toolchain packaged with the Arduino IDE it fixes ATtiny10 programming.
When the Arduino IDE was upgraded to the version 7.3.0 it stopped working like it was reported in the post.
Of course it works with the older versions of the Arduino IDE because it uses the 5.4.0 toolchain version.
It should just help people who develop on ATtiny10 and similar microcontrollers.

technoblogy, 2021-02-28:
These issues should now be solved by the new Boards Manager installation. Thank you for your suggestions.

closed 2021-02-28 by technoblogy