technoblogy/attiny10core #14: Compilation error on 8MHz clock of Attiny10 Blink example.

hmeijdam, 2021-04-08:
After I moved to version 2 of the core I get compilation errors on all my sketches including the example sketches when I select the 8MHz clock. The 1MHz clock setting compiles without errors

`
E:\Arduino\arduino-1.8.13\portable\packages\ATtiny10Core\hardware\avr\2.0.0\cores\tiny\wiring.c: In function 'init':
E:\Arduino\arduino-1.8.13\portable\packages\ATtiny10Core\hardware\avr\2.0.0\cores\tiny\wiring.c:8:3: error: 'CCP' undeclared (first use in this function)
   CCP = 0xD8;
   ^
E:\Arduino\arduino-1.8.13\portable\packages\ATtiny10Core\hardware\avr\2.0.0\cores\tiny\wiring.c:8:3: note: each undeclared identifier is reported only once for each function it appears in
E:\Arduino\arduino-1.8.13\portable\packages\ATtiny10Core\hardware\avr\2.0.0\cores\tiny\wiring.c:9:3: error: 'CLKPSR' undeclared (first use in this function)
   CLKPSR = 0b0000;
   ^

`
I am using a portable installation of Arduino IDE 1.8.13 on a Windows-10 PC

technoblogy, 2021-04-08:
Sorry about this; you need to add this to the start of wiring.c:

    #include <Arduino.h>

See:

https://github.com/technoblogy/attiny10core/pull/13

hmeijdam, 2021-04-08:
Confirmed that my compilation errors are gone now. Thanks for the fast fix.

closed 2021-04-08 by hmeijdam

technoblogy, 2021-04-08:
You're welcome. I'll try and get around to updating the library soon.