technoblogy/attiny10core #20: Programming via USBasp does not work using avrdude 6.3.0 (seems to only work with avrdude 7.3)

misaalanshori, 2024-06-26:
Hello I have been trying to program an attiny10 for a few days and this was the last problem I encountered. When uploading from Arduino, avrdude 6.3.0 is used and it does not work for programming the attiny10 using a USBasp.
[image]

I also tested multiple versions of avrdude: 6.3.0, 7.1, 7.2, and 7.3. Out of all of those versions, only 7.3 can reliably connect (read and write) to the attiny10
[image]
[image]

My current workaround is to just build the program using verify with verbose enabled, and then copy the .hex path to another terminal with the command to use avrdude 7.3 to actually upload it.

While that does work, it would be so much better if we could include avrdude 7.3 when installing the board into the Arduino IDE. I have tried to mess around with the package index file, but I honestly have no idea how to get it to retrieve and then use the newer avrdude 7.3 version.

technoblogy, 2024-06-26:
I'm using Arduino IDE 1.8.19 too with the default avrdude included with it, and last time I tried I didn't have any problems uploading to the ATTiny10, although that was a while ago. What version of firmware have you got on the USBasp?

misaalanshori, 2024-06-26:
I also thought the USBasp firmware was the issue, so I uploaded what I believe is the latest firmware (usbasp.atmega8.2011-05-28) to the programmer but ended up with the same errors.

technoblogy, 2024-06-26:
Are you programming the ATtiny10 in circuit, or a bare ATtiny10? I found programming was unreliable with other components connected to the ATtiny10.

technoblogy, 2024-06-26:
> While that does work, it would be so much better if we could include avrdude 7.3 when installing the board into the Arduino IDE. I have tried to mess around with the package index file, but I honestly have no idea how to get it to retrieve and then use the newer avrdude 7.3 version.

I think all you need to do is put the copy of `avrdude` you want to use in the ATtiny10 core at the same location as the `boards.txt` file, and change the line in `boards.txt`:

    attiny10.upload.tool=arduino:avrdude

to:

    attiny10.upload.tool=avrdude

misaalanshori, 2024-06-26:
> Are you programming the ATtiny10 in circuit, or a bare ATtiny10? I found programming was unreliable with other components connected to the ATtiny10.

I'm just programming a bare ATtiny10

> I think all you need to do is put the copy of avrdude you want to use in the ATtiny10 core at the same location as the boards.txt file, and change the line in boards.txt:

It looks like it still uses the avrdude from arduino (Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14) even after removing `arduino:`

technoblogy, 2024-06-26:
Did you restart the Arduino IDE?

Also, try adding:

    attiny10.bootloader.tool=avrdude

misaalanshori, 2024-06-27:
Yes I have restarted the IDE
[image]

I think it uses the tool versions defined in the index file?
[image]

technoblogy, 2024-06-27:
Sorry, I'm not sure then, and unless you can narrow the problem down I'm reluctant to change the ATtiny10Core because it may be a problem specific to something in your setup.

misaalanshori, 2024-06-28:
Yeah, it's very likely that it's caused by some weird quirk from my cheap USBasp board. The solution I have right now works well enough so it's fine I guess.

lern02, 2025-08-07:
I have the same problem as misaalanshori, Arduino 1.8.9. And in cmd administrator mode, when running avrdude -v, an error occurs:

         System wide configuration file is ""
avrdude: can't open config file "": Invalid argument
avrdude: error reading system wide configuration file ""

I have added the path to the Arduino IDE to the system environment variables.